simpleairquality/SimpleAQ-Device #55: Report:  Wifi Outages Require SimpleAQ Device Reset

theGOTOguy, 2023-10-02:
One of our users keeps having her wifi go out, and when it does the SimpleAQ device needs to be reset.

We should test by switching wifi off and on a few times and see if we can duplicate the issue.

theGOTOguy, 2023-10-06:
So, I noticed something odd.  I correctly set up a brand new SimpleAQ device and found that, in fact, on reboot not only did it not connect to the service but it's never even trying.

Commit 218bc0d461560da3a9aaa3d9266253332181a2c0 is investigating.

theGOTOguy, 2023-10-26:
I have duplicated the issue at 218bc0d.  That did not fix it.  The device is currently sitting, plugged in, stalled in my basement.

I  need to connect a monitor to this device to debug.

theGOTOguy, 2023-10-27:
Ok, I managed to catch this stall in action and connect to the stalled device.  A few hints.

Apparently, this happened after a reboot.  `dnsmasq` service failed to start, but is showing active (running) when I run `systemctl status dnsmasq.service`.  The final log message, shortly after data stopped reporting, was "failed to access  /run/dnsmasq/resolv.conf: no such file or directory".  The device never recovered and the config network is not up.

Examining the SimpleAQ service, it reports its last data packet and shows active (running), though no log messages have been reported since the final data packet.  It is apparently stalled somehow and not reporting any logs.

The very final log message was publishing SEN5X data, and [happened here](https://github.com/simpleairquality/SimpleAQ-Device/blob/main/simpleaq.py#L166).  GPS had no fix at this time.

I tried `sudo service simpleaq restart` to see if that resolves the issue.  It was not immediately able to connect to my local wifi, but is at least trying to do the right thing.  I will report back with my findings after I let it run for a while.

theGOTOguy, 2023-10-27:
Having run overnight, the unit did not reconnect to my local wifi.  I'm going to try `sudo service dnsmasq restart`.

theGOTOguy, 2023-10-28:
Ok, I checked the SimpleAQ service logs and it is indeed trying to bring the wifi back up.

I should manually try that command and see what happens, or what the result is.

theGOTOguy, 2023-10-28:
I stopped the SimpleAQ service entirely and instead ran the command SimpleAQ uses to bring up wifi:

`sudo systemctl start wpa_supplicant@wlan0.service`.

It appears to have started without issue.  Likewise, it brought down the `hostap` network.  

However, `curl www.google.com` still complains that we can't resolve the host.  

`nmcli` indicates that NetworkManager is not running (this might be correct,)
`iwconfig` indicates that we are indeed connected to the wifi.  However, oddly, I only see an ipv6 address listed here and not an ipv4 address like the other machines on my home wifi.

I wondered if it was just DNS and tried both `curl` and `ping` with 172.64.151.192 (www.slashdot.org) and got "network is unreachable".

theGOTOguy, 2023-10-28:
I checked the router:

[image]

Compare this to the results of ifconfig / iwconfig:

[image]

It looks like the MAC address for `wlan0` matches the IP address #2.  My suspicion is that 192.168.1.145 is my other, correctly functioning, outdoor SimpleAQ unit and that in fact the router doesn't even believe the device to be connected.

theGOTOguy, 2023-10-28:
For @bensonk, `dnsmasq` logs.  Nothing too interesting.  I think the file not found error is a red herring.

[image]

theGOTOguy, 2023-10-28:
[image]

`sudo service wpa_supplicant@wlan0 status` sure makes it look like I'm connected to my wifi, though.  60:38:e0:90:5f:21 is correct for my wifi router.

theGOTOguy, 2023-10-28:
`sudo arp -a` gives no output.

bensonk, 2023-10-28:
I feel like my current hypothesis is that while you're connected at the physical layer, and the data-link layer, you're not connected at the network layer (i.e. don't have an IP address). 

That would indicate that there was some problem getting a dhcp lease. I think the fact that you're using dnsmasq as the dhcp client further supports this theory, because I bet it's dhcp client featureset is somewhat less tested than, say, dhcpcd. And also less tested than its DNS serving capability. 

So, I guess for me the next step would be to try restarting dnsmasq, since it's the dhcp client, and see if it gets a lease. Also maybe set up a packet capture on your network (ideally at the point that serves DHCP) to see if you can see what DHCP packets it sends (or doesn't).

theGOTOguy, 2023-10-29:
That makes sense.  I checked `sudo service dnsmasq status` and found that it was crashed and couldn't bind 192.168.4.1.  I found [this comment](https://forums.raspberrypi.com/viewtopic.php?t=215235) and tried removing `bind-interfaces` from `dnsmasq.conf`.  That did allow me to restart `dnsmasq`.  

I am able to `sudo service wpa_supplicant@ap0 start` and see the configuration network as expected, but I am not able to connect when I run `sudo service wpa_supplicant@wlan0 start`.

So, I managed to get `dnsmasq` running but I'm not sure it effectively changed anything.

theGOTOguy, 2023-10-29:
I found [this issue](https://raspberrypi.stackexchange.com/questions/82547/wpa-supplicant-connects-to-wifi-but-gets-no-ip) which is pretty much exactly what's happening to us, right down to `wpa_cli -i wlan0 status`.  Their solution was to restart `dhcpd`.

In our case, restarting `dnsmasq` doesn't solve our problem.  But that's to be expected:  [We don't use `dnsmasq` on `wlan0`](https://github.com/simpleairquality/SimpleAQ-Device/blob/main/custom_pigen/stage2/04-install-requirements/files/dnsmasq.conf#L8).

Instead, we use [`systemd-networkd`](https://github.com/simpleairquality/SimpleAQ-Device/blob/main/custom_pigen/stage2/04-install-requirements/files/08-wlan0.network#L4) for `wlan0`.

The question is then:  how do I restart `systemd-networkd`'s DHCP service?

theGOTOguy, 2023-10-29:
`sudo systemctl restart systemd-resolved` did not fix the problem.

One possibility would be that we could use `wpa_cli -i wlan0 status` to search for the case where we're connected but can't get an IP and just reboot the system wholesale as a workaround.

theGOTOguy, 2023-10-29:
`sudo networkctl forcerenew wlan0` also did not fix the problem.

theGOTOguy, 2023-10-29:
`sudo service systemd-networkd restart` followed by `sudo service wpa_supplicant@wlan0 restart` did clear the issue.

It's noteworthy that it was important to remove the `bind-interfaces` line from `dnsmasq.conf`.

I need to figure out the best way to script this, but this should fix the rare issue.

theGOTOguy, 2023-10-30:
Ok, https://github.com/simpleairquality/SimpleAQ-Device/commit/10d99b947f0d97236496a0388beac314c693dbbb should make this type of error recoverable.

theGOTOguy, 2023-10-30:
I will update this after I've tested the image.

theGOTOguy, 2023-10-30:
Ok, got the device flashed and testing.  A couple of comments.

1)  After first boot, I didn't notice the wifi config network come up.  I went over and logged into the device, ran a couple of `service status` commands, didn't see any issues, came back and there it was.  It's possible I didn't notice it on the list.
2)  It configures and reports data fine.

I'm going to let it run for a few days and try periodically blocking it from wifi to see if I can reproduce a failure.

theGOTOguy, 2023-10-31:
Ok, I put the firmware in another device and indeed it looks like this time the config network came up.  I must have missed it.  I'm going to start this onw running too to see if any issue reproduces.

theGOTOguy, 2023-10-31:
I was going to set up the second device and noticed that the other one has disconnected from my wifi and its configuration network came up.  It must be a weak connection in that spot.  I'll wait and see if it reconnects before setting up the second device.

theGOTOguy, 2023-10-31:
Ok, the original device did indeed come back up.

theGOTOguy, 2023-11-01:
Both devices still going strong.

theGOTOguy, 2023-11-02:
Still good.  I'm going to intentionally put one in a Faraday bag for a while, then take it out and see if it comes back up.

theGOTOguy, 2023-11-02:
Well, it turns out this [wifi/RFID blocking bag](https://www.amazon.com/dp/B08XMF32CM?psc=1&ref=ppx_yo2ov_dt_b_product_details) doesn't work at all.

theGOTOguy, 2023-11-17:
I believe this may be fixed by https://github.com/simpleairquality/SimpleAQ-Device/commit/10d99b947f0d97236496a0388beac314c693dbbb -- there is some issue with systemd-networkd that causes it to fail if you keep bringing interfaces up and down.  We have worked around it by rebooting it when we bring an interface up.

closed 2024-12-11 by theGOTOguy