simpleairquality/SimpleAQ-Device #56: Adafruit I2C GPS is Discontinued

theGOTOguy, 2023-11-17:
[The Adafruit I2C GPS we supported is now discontinued, and the new V3 is USB only](https://www.adafruit.com/product/4279).

We will have to:

1. Redesign our hat to support the new V3 (or whichever GPS we decide to support going forward,) 
2. Update our firmware to automatically select between the legacy and the new USB GPS.
3. Update our build instructions.

For now, I am going to buy out all of the remaining stock from Digi-Key.

theGOTOguy, 2023-11-17:
Update:  I am ordering one for testing.

bensonk, 2023-11-17:
Well, darn. I'm not sure if USB is the right play, I think maybe we should go to a UART gps instead. It'll probably require about the same amount of nonsense to connect it, and we can get much much cheaper GPS units.

theGOTOguy, 2023-11-17:
Sure, link me whatever product you prefer and we can standardize on that instead.

theGOTOguy, 2023-11-17:
These may be options for us:

[Flashtree BN-180](https://www.flashtree.com/products/flashtree-bn-180-gps-module-uart-ttl-dual-glonass-gps-car-navigation-gps-antenna-for-arduino-raspberry-pi-pixhawk-aircraft-cc3d-f3-flight-controller?variant=34051067969595&currency=USD&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&srsltid=AfmBOorj6tmtdUDmvzcWXBGiS7wn9uINqNfeSNbjAyafb1IbSUynff54j0U)
[SparkFun GPS Breakout](https://www.sparkfun.com/products/15193?gclid=CjwKCAiAu9yqBhBmEiwAHTx5p2hYQvlIxLCYZq3skDiO8VkL1ce5s2YYuEX3dhCk9rmL_C3-sBIU5BoC9JUQAvD_BwE)
[DFRobot GNSS/GPS](https://www.google.com/shopping/product/12506108989592518222?sca_esv=583408504&sxsrf=AM9HkKkD7fI7UQjRKekDu0y5LTgEP27XqQ:1700247192052&q=i2c+gps+module&sqi=2&biw=3839&bih=1966&dpr=1&prds=eto:341296748726157504_0,pid:8334733738544551170&sa=X&ved=0ahUKEwiD_oq42suCAxUXkIkEHZlRBmsQ8wIIiA0)

theGOTOguy, 2023-11-28:
Dr. Hood wants 20 more units.  I'm going to buy a couple more to evaluate.

I do have the new Adafruit USB one, but it's expensive, awfully large, and I agree with @bensonk 's concern that USB isn't the right play for us.

theGOTOguy, 2023-11-28:
Ok, I'm going to try the DFRobot GNSS/GPS.  

Pros:
- It's half the price of the Adafruit units.
- It has Python drivers: https://github.com/DFRobot/DFRobot_GNSS

Cons:
- It uses a different connection standard versus Adafruit (DFRobot's versus Adafruit's connector.  Maybe we can find an adapter?)
- It has a different form factor and we'll need new hats.

And a question:
Since you can't even buy Adafruit's GPS anymore, is it reasonable to drop support for it?  Or should we code in the possibility of different GPS units?

bensonk, 2023-11-28:
If we drop support for it, won't all the existing units break?

theGOTOguy, 2023-11-29:
If we drop support for the old GPS units, it will not affect existing units
unless they update their firmware.

theGOTOguy, 2023-12-17:
Commit 3a1913e adds prototype support for DFRobot's GPS, while continuing to allow legacy support for the old Adafruit PA1010D.

If it works pending field testing, I like a lot of things to like about the DFRobot GPS.

1. The lower price.
2. Doesn't use any non-standard Python libraries.
3. Also provides altitude and number of satellites used.
4. Allows multiple modes, separately or in concert:  GLONASS, GPS, and BeiDou
5. No asynchronous processes leading to intermittent data corruption, something we actually saw in the Adafruit one.

In fact:  we can even support multiple different GPS units attached at once!  The behavior will be a race, but it will work.

theGOTOguy, 2023-12-31:
#57 Adds support for DFRobot's I2C GPS unit, but I think we're actually going to move to UART GPS support instead.

theGOTOguy, 2024-01-01:
Based on the recommendation of @bensonk , I bought a UART GPS:  https://www.amazon.com/dp/B078Y4XZN9/?th=1

theGOTOguy, 2024-01-28:
This is now done, save for updating build instructions

closed 2024-12-08 by theGOTOguy