simpleairquality/SimpleAQ-Device #69: Evaluate Current Board Revision

theGOTOguy, 2024-12-08:
In the current board revision, Tyler has kindly added a 4k-ohm pull-up resistor.

We should evaluate:
1)  Whether this is sufficient.  I am still seeing some bus flakiness, but it seems to recover most of the time.  https://forums.adafruit.com/viewtopic.php?t=187016 suggests that we could even use a 1k-ohm resistor if we wish to be more aggressive.
2)  Whether we wish to add a MOSFET to GPIO pin 17 and the 5v supply to GPS and I2C so that we can soft-reset the devices on the bus if desired.  I want to try a software update before we go down this road, as I have seen the original Board + GPS + BMP + SEN5X sensors run for months at a time with no issue.

@bensonk I have measured the capacitance of the I2C bus on Dr. Hood's EPA device.  400pF is I2C's design limit.

This is a cheap Chinese LC meter, so take each of these readings with a grain of salt.  They might be correct in an order-of-magnitude sense.  Readings are for SDA-GND
23 pF:  Probes only
1.77 nF:  BMP390 only (on its face, this is already higher than the allowable 400pF on a 100kHz standard I2C bus!)
3.61 nF:  Board only.  No devices attached.  
3.57 nF:  Board + GPS.  The GPS shouldn't affect this reading, but I am adding it as a control.
5.30 nF:  Board + GPS + BMP390
5.41 nF:  Board + GPS + BMP390 + SEN5X
7.15 nF:  Board + GPS + BMP390 + SEN5X + 1 DFRobot Gravity MultiGas
8.88 nF:  Board + GPS + BMP390 + SEN5X + 2 DFRobot Gravity MultiGas
10.61 nF:  Board + GPS + BMP390 + SEN5X + 3 DFRobot Gravity MultiGas

So, either my meter is nuts or I'm missing something.  On its face, even just the BMP390 alone should prevent I2C from working, but we've seen it work for 8 months straight with no hiccups.  If you used the [official guide](https://www.ti.com/lit/an/slva689/slva689.pdf?ts=1733639117315&ref_url=https%253A%252F%252Fwww.google.com%252F), we need an 0.11 ohm pull-up, below the allowable minimum size.  In other words this isn't even supposed to be possible on I2C.  

What I'm taking from that document is that we could actually use a much more aggressive pull-up resistor, even in the hundreds of ohms, to try to stabilize the bus.

theGOTOguy, 2024-12-08:
Update:  I tried the capacitance of a different BMP390 and got a very similar value.

Another thing we could try:  [Active pull-ups](https://www.edn.com/i2c-active-pullup-is-power-thrifty/) are apparently able to boost the max allowable capacitance into the nF range.

theGOTOguy, 2024-12-11:
I have confirmed that we are able to deliberately cause bus stuckness by repeatedly crashing the SimpleAQ service without gracefully closing the I2C connections.

theGOTOguy, 2024-12-12:
I'm increasingly pessimistic that this is going to be sufficiently stable with the 4k pull-up, but I'm still making a lot of changes to the software to improve things on that side.  Tyler is working on switching us over to an LTC4311-based active pull-up: https://www.analog.com/media/en/technical-documentation/data-sheets/4311fa.pdf

theGOTOguy, 2024-12-14:
Ok, so I just saw a device on my desk boot-loop for 10 minutes and I couldn't figure out why.

I think it's because the voltage regulator on there has an automatic thermal shutdown, and it was getting too hot.

I put a heatsink on there and that stopped the issue, but even this heatsink gets hot.  

We have two options:
1)  A heatsink to disperse heat, or
2)  Reducing the voltage of our power supply from 9V to, say, 7V.  This will give us less power but also dissipate less heat, or
3)  Tyler has suggested that we could use USB-C power and let the wall wart figure it out.

theGOTOguy, 2024-12-25:
So, I haven't ordered or tried Tyler's new board revision yet, but I'm evaluating the older revisions of the board.

One thing I notice is that the bus issues always  unfold the same way:

- OS Errno 5 (Input/output) error when reading data from a DFRobot Multi-Gas device
- Hang forever when publishing BMP3XX data.

I found this, indicating that the BCM2835 chip has a bug with no workaround or fix except "don't use any devices that utilize clock stretching," which will cause bus stuckness when clock stretching:  https://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html

There is an option.  We could switch to [software i2c](https://github.com/fivdi/i2c-bus/blob/master/doc/raspberry-pi-software-i2c.md), which would use different pins.  

Or, we could experiment with a Raspberry Pi Zero 2W, which uses a BCM2710A1.

theGOTOguy, 2024-12-26:
So, I'm noticing a few more weird things:

All of SimpleAQ service seems to hang on one of these lines during bus stuckness:
- https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/main/adafruit_bus_device/i2c_device.py#L172
- https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/main/adafruit_bus_device/i2c_device.py#L153
- https://github.com/adafruit/Adafruit_CircuitPython_BMP3XX/blob/main/adafruit_bmp3xx.py#L147

So far we've been stable for 7 hours with this build:  https://github.com/simpleairquality/SimpleAQ-Device/actions/runs/12496783549

But, I noticed something:  in this particular case, we publish the BMP3XX data before the DFRobot devices in the order of publish.  I wonder if this matters.

theGOTOguy, 2024-12-26:
Ok, and moving BMP3XX to the middle of the devices seems to have caused an infinite loop at "waiting for read_byte in adafruit bmp3xx" almost immediately.

theGOTOguy, 2025-01-29:
So, I am writing at 4:30AM and I apologize that I do not have a properly formatted USB drive to save screenshots from my oscilloscope.

With the new Rev 1.41C boards, I started by using a board that had all pull-up resistors and the active pull-up chip removed, as well as the capacitors.  This is basically just connecting USB-C power to everything.  I have connected only a SEN5X.

- Attaching to the "BMP" port, it looks like SCL is normal, with a waveform that looks good.
- I can't seem to read much of anything off of SDA, and when I do, it looks nothing like the expected digital shape.  It looks like noise, and not necessarily digital.  Smooth, narrow peaks and valleys not always reaching to 0V or 5V.

Will continue to report findings as I test further.

theGOTOguy, 2025-01-29:
The single DFRobot plus the SEN5X seem to work the same regardless of what port I plug the DFRobot into.

theGOTOguy, 2025-01-29:
With two DFRobots and this SEN5X on the board with all non essential parts removed, we detect nothing and also see a constant few hundred mV on SDA.  

Clearly there is not enough power on the bus in this case.  GND-PWR reads 5.1V so the USB power supply does appear to be sufficient.

theGOTOguy, 2025-01-29:
So I went back to the full 1.41C board.

Plugged in the SEN5X only, and the SEN5X works.  One thing I am noticing, both on the board with parts removed and this one, is that the high level plateaus on SDA and SCL are always 4V as opposed to 5V.  

You can actually see the I2C extender switching on and dumping the full 5V rail into SDA/SCL, then it dropping down to 4V due to some internal resistance.

[image]

theGOTOguy, 2025-01-29:
Adding in the BMP3XX, it does work, but the GPS stopped working.  Then I rebooted and the GPS worked but the BMP did not.  Waveform still looks good, within the plateau voltage being low-ish.

I am starting to wonder if this is simply a power supply issue.

[image]

theGOTOguy, 2025-01-29:
As I think about it: I wonder if the voltage we are dumping into SDA/SCL when the I2C extender switches on is causing a drop on the 5V rail that is causing the GPS to fail.

theGOTOguy, 2025-01-29:
Yeah, and here we are with 3 devices on I2C.  Flatlined.  This is a power issue.

The question is: do we need a higher voltage regulated supply, or just lower resistance pull up resistors?

[image]

theGOTOguy, 2025-01-29:
Even with an official Raspberry Pi power supply (5.1VDC 3A) vs the cheap one I had been using (5VDC 1A), I am still seeing the flatline with 3 devices on the bus.  The GPS does come up.

theGOTOguy, 2025-01-29:
OK so I went back to the old revision (with the reverse cables) for the sake of argument.  

Even with all devices plugged in, it does come up and apparently work, except the BMP (do I misremember and that one didn't need a reverse cable?)

Waveform plateaus are still at 4V, but don't flatline with "too many" devices attached.

[image]

theGOTOguy, 2025-01-30:
Ok, so for whatever reason I can't get the BMP to register anymore, but I swear it did in the past.

I'm actually wondering, looking at this silkscreen, if it's @MetalCard's knockoff Alibaba BMP390.  I might have swapped it when I was working on another device.

So, here's my plan.

1)  I am going to order a known-good Adafruit BMP390.
2)  I am going to order a breadboard and resistors from Adafruit.  I should be able to simulate stronger passive pull-up resistors by connecting one of the Gravity pins to a breadboard, then adding different resistances between 5V and SDA/SCL.  I suspect this will raise the "plateau" on SDA/SCL, which may improve the signal well enough to get the current design to work within better passive pull-ups.

What do you think, @MetalCard?

theGOTOguy, 2025-02-08:
I frankensteined together 1K pull-ups in parallel with the existing 10K pull-ups.  This is with everything plugged in  3 Gravity devices, BMP, SEN and GPS.  Note the plateaus at 5V on the nose.  

[image]

theGOTOguy, 2025-02-08:
And yet:  Nothing at all is detected except uartnmeagps.  Maybe I made a wiring error?

theGOTOguy, 2025-02-08:
I removed the BMP390 from the unit, rebooted, and flatline on SDA/SCL with no devices detected.

Something about that BMP390 hardware must be affecting the bus.  Maybe it has onboard pull-ups or something?

MetalCard, 2025-02-08:
> I removed the BMP390 from the unit, rebooted, and flatline on SDA/SCL with no devices detected.
> 
> Something about that BMP390 hardware must be affecting the bus. Maybe it has onboard pull-ups or something?

Yep, all the external components have pullups on them as they assume none would otherwise be present since they are designed to directly attached to a Pi, ESP, Arduino, etc.

theGOTOguy, 2025-02-08:
Ok, I was continuing to see weird effects with the new board so I decided to go back to basics.  I just want to see the waveforms and whether it works.

Last iteration board, reverse cables on everything, and the only real difference being that I'm using the known-good BMP3XX.

I'm going to reboot it a few times to confirm that everything always comes up.

Notice that the known-good BMP3XX (black connectors) versus the possibly AliBaba Knockoff (white connectors) does look different, and also the silkscreen is subtly different.  

[image]

theGOTOguy, 2025-02-08:
Everything came up correctly first try with the known-good BMP390.  Waveforms look pretty crisp, within the rise time:

[image]

theGOTOguy, 2025-02-09:
@MetalCard has pointed out that neither of those BMP390s are the Alibaba knockoff, which had a purple PCB.  It's possible that I damaged the flaky one when I hooked it up in reverse earlier.

theGOTOguy, 2025-02-09:
OK now I'm testing with the new revision, but with no pull-ups (active or passive) and no caps.  Only the indicator light.

1)  Works with SEN5X (only)

theGOTOguy, 2025-02-10:
Ok, I frankensteined in aggressive 2K passive pull-ups on SDA/SCL.  

1)  Works with SEN5X (only).  Note crisp, 5V peak-to-valley waveforms.

[image]

2)  Works with SEN5X and UART/NMEA GPS.

theGOTOguy, 2025-02-11:
So, I wanted to test two questions to make sure we fully understand what's going on.  I rebooted 3 times and it comes up fine every time.

1)  Does this work with the other, older BMP390?  Was that a red herring, and
2)  Does the new, unmodified board work with this configuration?

theGOTOguy, 2025-02-11:
Trying with the other BMP390, it doesn't seem to detect it.  It's possible that wiring that one backwards damaged it.  Works fine as soon as we put the known-good BMP390 in.

theGOTOguy, 2025-02-11:
And, taking all of the stuff and plugging it into the new board (plus my frankenstein 2K pull-up resistors):  It works on the first try.

I'll make sure that it does for a few reboots.

Now I have question for the panel.

@bensonk @MetalCard 

The new board, including capacitors to stabilize ripple and active pull-ups, apparently works fine once we add 2K pull-up resistors.   It turns out, it also works fine with only 2K pull-up resistors, no caps, and no active pull-ups at all.

So:  which is the design we go forward with?  The one with the fancy active pull-up and caps, or just a vanilla build with 2K passive pull-ups instead of the 10K we have now?

bensonk, 2025-02-14:
What happens when you add additional sensors? Is there a way to test whether the additional caps and active pull ups are helpful in adverse conditions?

theGOTOguy, 2025-09-17:
Closing, I'm reasonably happy with the present board revision.

That said, we have seen bus stuckness and if there's a solution, it's this:  https://github.com/simpleairquality/Device-Models/issues/1

closed 2025-09-17 by theGOTOguy